nicsure/QuanshengDock #1: can't use CHRP

snipsnap17, 2024-01-08:
Have two radios, on both supplied with your fimware, it is't possible to use Chirp.

nicsure, 2024-01-08:
It shouldn't affect CHIRP as the firmware is essentially just egzumer, it doesn't mess with the eeprom flashing or anything. I'll look into why this is happening.

victor7376, 2024-01-08:
Just tried Chirp & it states 'bad header' so cant download or upload from the radio

nicsure, 2024-01-08:
Interesting, That's kind of surprising. I'll run CHIRP myself and try to see what the problem is.

nicsure, 2024-01-08:
Oh I know what's happening. The radio is sending display information constantly, that's what's confusing CHIRP. I can sort that out.

nicsure, 2024-01-08:
I've made a temporary solution for now. It requires reflashing the firmware with the release I've just updated on the firmware page. In the radio menu you'll find a new option called "Remote", if you set this to "OFF" CHIRP should then work. This setting defaults to "ON" every time the radio is powered up or reset.

ErikS-web, 2024-01-10:
Nice solution :+!: 

Suggest to correct title and add the "I", before closing, this issue as solved.

ErikS-web, 2024-01-10:
Where is this new firmware published ?

I only see release 1 firmware at [releases ( 4 days old) ](https://github.com/nicsure/quansheng-dock-fw/releases)

nicsure, 2024-01-11:
I didn't actually make a new release, just replaced the bin file.

ErikS-web, 2024-01-17:
Solution work fine:

Menu 60.`Remote` is default ON (after power-up). 

And can be set to "OFF" before using Chirp.

Just one question/suggestion: 
_Could it be made possible to change this parameter to OFF from PC  in QS-Dock ?_

Than..._This item can be closed ;-)_

closed 2024-01-22 by nicsure