nicsure/QuanshengDock #4: ChDisp (menu 33) on "Name + Freq." but not in app display

ErikS-web, 2024-01-08:
_**compliments**_ for this nice idea. :+1:

I use setting `ChDisp` = **Name + Freq.**

Only _both frequencies_ are on screen and two rare stripes, on the place for the Channelname

De MAIN-channel has a shorter stripe. Looks like the ► overwrites the name

[image]

and some more 'cosmetic' findings
[image]

- No DW on top-line but an **UWR** ?
- **Battery logo** is empty ?
- **Voltage** is not exact equal.

nicsure, 2024-01-08:
Try a different font and see what happens, the > mark is a unicode character so it may be 
casuing issues on your system.

I'll take a look at the status options, I may have mislabelled something somewhere.

The battery is just a character, it doesn't "fill up" like that. I decided to just show voltage and percentage.

Yes, I know about the battery voltage readout, the reason is because the firmware sends the battery data to the PC as 8 bits, but it's actually a 16 bit value so the value is too course. It was a bad choice and it's something I'm going to address.

nicsure, 2024-01-09:
I indeed had mislabed dual watch and respond. Now changed to "DWR".

ErikS-web, 2024-01-09:
Using latest build: https://github.com/nicsure/QuanshengDock/commit/d84796d4452f5a7baf4e092e4b55df131a1629c7

1. Yes, but why the "R" ?

That R is not on the display of the radio.

Also when doeing TX/RX there is no TX/RX in front of the "channel name". But you asked me to out some settings.
Changing font was a disaster. And gave big characters over character.

- But when adjusting `Fstretch` from 1,17 to 1,00 picture looks OK

- Now on channel B power Low but I downt see the horizontal line of the "L", but this erro also on channel A.
    Must be the font. It is the default "Segoe UI"

[image]

- Also see under the Numbers of the squelch values.

   Parts of previous number are visuable under `107 S3`

ErikS-web, 2024-01-09:
Do you mind if I continue here with my findings. Or do you prefers new issues ?

nicsure, 2024-01-09:
It's DWR because it's called "Dual Watch and Respond", also the bitmap for it in the firmware is DWR, I'm not sure why the display only shows DW tbh. I'll look into that further, the status line is reconstructed and quite complex. Leave it with me.

There seems to be some issue with font rendering, it's likely locale related. I've seen similar font problems elsewhere. There was a Polish guy showing the software on a video and his display had no numbers at all. So it might be down to the language/culture settings of the Windows installation.

nicsure, 2024-01-10:
From what I can tell the majority of your issues are related the F Stretch function and it being applied differently on some installs. So I changed the way it works, it's no longer a scaler value. Well, it kind of is, but instead of being multiplied by the V Size as it was before, they are now just added together.

ErikS-web, 2024-01-10:
F-stretch does not make my screen crazy 👍 

On recieving there is also the R in the left-above corner as is on the radio.
But with TX there is no **T** on the Dock-screen as is on the radio.

ErikS-web, 2024-01-10:
Mmmhhh and now it is OK ???

I will look into this tomorrow again...

ErikS-web, 2024-01-12:
In VFO mode the characters are to BIG

See the 000 under the 225

[image]

**Tip: _make a version number visual on the screen_.** 
If an issue is reported with a print screen, we can immediately see the version used

ErikS-web, 2024-01-12:
Here is the screen of the issue from yesterday.

No T in upper left corner on TX. And also NO RED LED-indicator.
Today also this issue.

[image]

On RX all OK. Green LED and also the R

ErikS-web, 2024-01-14:
The Red-LED for TX works fine, when channel-B is used as primaty-channel.
So only upper VFO (a) fails with no Red-LED on TX.

nicsure, 2024-01-16:
Your text is too big, f stretch should be around 0.2 ish

TX for me is working fine, cannot replicate.

ErikS-web, 2024-01-17:
Did find my TX-error. Also HF-interference at some specific 145 Mhz. Yep metoo...

I think it is cause bij the cabling, specific because the 3,5 mm extension has NO GROUND wire.

I am thinking on how 2 change and improve the wiring and specially the shielding.

ErikS-web, 2024-01-17:
F-stretch is now with value 0.80 Nice screen

[image]
[image]

But look here...

* what are the 3 `???` in topline

* yesterday I save 145.725 and 145.125 as Repeater, but look what is now on screen
* No repeatername bur number ?
* second memory also false and very strange
[image]

* I can delete-and-insert... but new Idea... where is the EDIT / Modify ?

nicsure, 2024-01-17:
Probably version mismatch on the preset data, I try not to do that but it's hard to prevent it sometimes. The ??? means that the preset has a bandwidth value it doesn't recognize.

To edit or modify a preset you use "Set to Current" which updates a preset to the current VFO config. So if you want to change something on a preset, load it in, make the change on the VFO then select "Set to Current" on the same preset.

closed 2024-01-23 by ErikS-web