nicsure/QuanshengDock #5: CAT worked with Digirig

TavaDelphin, 2024-01-09:
Woot!

I was going through the setup and having issues with an errant USB driver (unrelated to the Quansheng Dock), when I decided to try to use the cables I got with my Digirig to see what I could make work.

Initially, I found very limited functionality but when I connected the radio to the Digirig with the green end cables to the Serial port I suddenly had sound and PTT functionality, a little more configuring and I was broadcasting voice over the air - VICTORY LOL - followed shortly thereafter with a Microsoft error and the app shutdown. 

I have repeated it all including varied crash messages.   

That being said it does work after a fashion allowing full computer control and voice through the headset.  Great program with superior potential.

If I can get some pattern to the crashes I will forward the details to you,   
Bruce  
WROE975

nicsure, 2024-01-09:
The errors could be due to TX energy leaking into the serial cable line, I had similar problems during development while using the supplied antenna. It all went away when I switched to an external antenna in the loft. Not saying that's the problem but it's something to consider.

TavaDelphin, 2024-01-09:
Excellent I can do that too and will advise. My programming skills have atrophied to the point of non-existence so I can offer little constructive support but will happily play and report what I find. Bruce CarrFrom iPhone

betzburger, 2024-01-20:
Wow, QuanshengDock is a great idea. I also own a Digirig with the Baofeng cable. Does this work or do I need to build this cable mod?

TavaDelphin, 2024-01-20:
I used the green digirig cable initially and it worked fine. Unfortunately I have had issues with my digirig recently and can not validate it still works as desired - sorry.

TavaDelphin, 2024-01-20:
Shoot sorry I used it in the serial side too. The other cable did work for listening too.

nicsure, 2024-01-20:
I currently have a hacked together bluetooth arrangement. Although the audio is not very good, the bluetooth sender is fine in simplex, but when in duplex it shifts to low sample rate mono. Looking for a better bluetooth sound interface.

ErikS-web, 2024-01-30:
To me Digirig is also an **All In One Computer** interface.

Is this Issue a **_software problem_** [BUG] or just a [TEST] case of some other peripheral possibility.

Suggest to adjust title with [TEST] and consider to add it under [options-for-connectivity-that-are-being-investigated](https://github.com/nicsure/QuanshengDock/wiki/30-%E2%80%90-Connection-between-radio-and-PC#options-for-connectivity-that-are-being-investigated)

And if you think this is interesting ... consider [support](https://github.com/nicsure/QuanshengDock#support)

TavaDelphin, 2024-01-30:
HiNot sure how to properly add an entry to the page.  I have not used GitHub, except to try to make a custom firmware setup - which failed lol. I would be happy to share my info on the Digirig. As an aside may Digirig is down, not sure what I did to it but it emits a loud single tone when powered on now. But it did work great for a few iterations of the software. Bruce CarrFrom iPhone

ErikS-web, 2024-01-31:
Very simple... just login Github and select your/this issue.
Scroll to below.... to 
**Add a comment**

And to the right below an option `Close issue`.

I suggest to close it for now. [I have added Digirig as an option to investigate, in the Wiki](https://github.com/nicsure/QuanshengDock/wiki/30-%E2%80%90-Connection-between-radio-and-PC#digirig)

closed 2024-02-02 by TavaDelphin