nicsure/QuanshengDock #6: PTT not working with AIOC

tristanmoney, 2024-01-09:
Really neat software, but PTT doesn't seem to work with a AIOC. Everything else works though.

I think the AIOC is the perfect device to pair with this software.
https://github.com/skuep/AIOC

The AIOC also has a transistor to assert the PTT, so maybe that could be used instead of a serial command?

nicsure, 2024-01-09:
I'd be more than happy to test the AOIC out if they'd send me one. As I don't own one I can't really support it.

tristanmoney, 2024-01-09:
Please email me your address, I will order you some AIOC 's from JLCPCB.

nicsure, 2024-01-09:
Thank you very much for that offer Tristan, I really appreciate your generosity. I've already had a viewer contact me and has sent one to me. Thank you again for that offer though, it really does mean so much to me that this community is so supportive.

nicsure, 2024-01-09:
Just to be sure, you are unlocking the TX on the Dock interface aren't you? The top right button?

tristanmoney, 2024-01-10:
I didn't know about that button! TX works fine once unlocked. Sorry I only
had 15 minutes to try it out, please close the issue.

closed 2024-01-10 by tristanmoney

ErikS-web, 2024-01-14:
When the AIOC is plugged into the radio, can you listen to the audio from the radio ?