nicsure/QuanshengDock #10: App crashes after reading from radio with Channel edit.

k30lls, 2024-01-10:
Hi, all worked fine after running firmware and downloading the dock app.
I had to disable Norton as it was detecting a virus.
When i read from the radio, it reads fine but when i scroll down to view the channels, it crashes the app.
could be something my end.

nicsure, 2024-01-10:
Norton? Well I'd have to question your choice of security software there I'm afraid. Unfortunately I cannot control what Norton may or may not think is malicious, all I can do is make it open-source so everyone can inspect it for things like that. I suspect that Norton just flags any unsigned app it's not familiar with, it's not a particularly good security suite and uses a whole lot of lazy techniques.

You may have some erroneous values in the channel data on the radio that's upsetting the app, I'll add a more strict sanitizer of the data read from th3e radio. I'm working on the channel editor at the moment as I'm adding RepeaterBook integration.

k30lls, 2024-01-10:
Here you go. Only saved a few test channels on the radio directly to try.
Had to zip as wouldn't accept file type.
[test channels.zip](https://github.com/nicsure/QuanshengDock/files/13892020/test.channels.zip)

ErikS-web, 2024-01-24:
Any improvement with latest version of Dock (and firmware) ?

ErikS-web, 2024-01-27:
Must be OK than. Closing Issue.

closed 2024-01-27 by ErikS-web