nicsure/QuanshengDock #12: Alternative to Hardware Configuration

ErikS-web, 2024-01-10:
That microphone has an earplug at the bottom.

The radio audio also comes from there. You do have 2 cables to your microphone, but it is just as easy to make.

Possibly add a resistor internally and possibly also a capacitor due to DC voltage from the UVK5(8) on the audio-signal

[image]

ErikS-web, 2024-01-10:
An Y-cable for 3.5 is easy to find at Ali....

Any suggestions for the 2.5 mm Y-cable ?

nicsure, 2024-01-10:
Well throw me in a bucket of custard. I had NO IDEA the microphone had a ear port on the bottom. LOL!

ludwich66, 2024-01-20:
I had opened this kind of handset - a draft picture

[image]

ludwich66, 2024-01-20:
Please Verifiy and Comment Schematic:
There is Version A (First Cable connection) and
Version B (Connect Soundcard over Special-Cable to External-Handset

[image]

ludwich66, 2024-01-20:
Please check this Schematic:
Version 1 : Old QS-Dock-Cable
Version2 : Connection over Handset-Earphone-Connector
[image]

ErikS-web, 2024-01-22:
Another idea: 

You can also use _a second 2.5 mm Y-cable_/Split cable to split the audio to the PC.

And then insert the resistor into the 2.5mm plug

ludwich66, 2024-01-22:
[image]

ErikS-web, 2024-01-22:
Where to buy in 2.5 mm ?

closed 2024-01-24 by ErikS-web