nicsure/QuanshengDock #8: AIOC cable compatible?

waydenie, 2024-01-10:
Wondering if this was used with an AIOC device ( https://github.com/skuep/AIOC ) , would this be able to use usb audio devices on the PC , instead of the audio patch wiring?

nicsure, 2024-01-10:
I have an AIOC on the way, so it should be supported soon hopefully.

ErikS-web, 2024-01-10:
See (closed) issue #06

waydenie, 2024-01-10:
Ah! thanks so much, I missed that one...

I finally found a few mins to find my AIOC, install the firmware and run the app with my AIOC. Runs great, as described in issue #6! thanks for this great software, I think this issue can be closed too!

ErikS-web, 2024-01-10:
You should be able to close your issue yourself... See below

waydenie, 2024-01-11:
Thanks again, this is a great project!

closed 2024-01-11 by waydenie