nicsure/QuanshengDock #9: Buffer full exception, F+5 not launching spectrum, F Stretch not working

aliatx2017, 2024-01-10:
1. Get a .NET exception while receiving audio (latency 40, buffer 45).
2. Pressing F+5 should launch spectrum analyzer, currently does nothing on the radio.
3. Lowering F Stretch value in settings, closing settings and pressing the Exit button to apply doesn't adjust width

[image]

[image]

[image]

See the end of this message for details on invoking 
just-in-time (JIT) debugging instead of this dialog box.

************** Exception Text **************
System.InvalidOperationException: Buffer full
   at NAudio.Wave.BufferedWaveProvider.AddSamples(Byte[] buffer, Int32 offset, Int32 count)
   at NAudio.Wave.WaveIn.Callback(IntPtr waveInHandle, WaveMessage message, IntPtr userData, WaveHeader waveHeader, IntPtr reserved)
   at NAudio.Wave.WaveWindow.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, WM msg, IntPtr wparam, IntPtr lparam)

************** Loaded Assemblies **************
System.Private.CoreLib
    Assembly Version: 6.0.0.0
    Win32 Version: 6.0.2623.60508
    CodeBase: file:///C:/Program%20Files/dotnet/shared/Microsoft.NETCore.App/6.0.26/System.Private.CoreLib.dll
----------------------------------------
QuanshengDock
    Assembly Version: 1.0.0.0
    Win32 Version: 1.0.0.0
    CodeBase: file:///C:/temp/quansheng/dock/QuanshengDock.dll
----------------------------------------
PresentationFramework
    Assembly Version: 6.0.2.0
    Win32 Version: 6.0.2623.60503
    CodeBase: file:///C:/Program%20Files/dotnet/shared/Microsoft.WindowsDesktop.App/6.0.26/PresentationFramework.dll
----------------------------------------
WindowsBase
    Assembly Version: 6.0.2.0
    Win32 Version: 6.0.2623.60503
    CodeBase: file:///C:/Program%20Files/dotnet/shared/Microsoft.WindowsDesktop.App/6.0.26/WindowsBase.dll
----------------------------------------
System.Runtime
    Assembly Version: 6.0.0.0
    Win32 Version: 6.0.2623.60508
    CodeBase: file:///C:/Program%20Files/dotnet/shared/Microsoft.NETCore.App/6.0.26/System.Runtime.dll
----------------------------------------
System.Xaml
    Assembly Version: 6.0.2.0
    Win32 Version: 6.0.2623.60503
    CodeBase: file:///C:/Program%20Files/dotnet/shared/Microsoft.WindowsDesktop.App/6.0.26/System.Xaml.dll
----------------------------------------
System.IO.Packaging
    Assembly Version: 6.0.0.0
    Win32 Version: 6.0.2623.60508
    CodeBase: file:///C:/Program%20Files/dotnet/shared/Microsoft.WindowsDesktop.App/6.0.26/System.IO.Packaging.dll
----------------------------------------
System.Private.Uri
    Assembly Version: 6.0.0.0
    Win32 Version: 6.0.2623.60508
    CodeBase: file:///C:/Program%20Files/dotnet/shared/Microsoft.NETCore.App/6.0.26/System.Private.Uri.dll
----------------------------------------
PresentationCore
    Assembly Version: 6.0.2.0
    Win32 Version: 6.0.2623.60503
    CodeBase: file:///C:/Program%20Files/dotnet/shared/Microsoft.WindowsDesktop.App/6.0.26/PresentationCore.dll
----------------------------------------
DirectWriteForwarder
    Assembly Version: 6.0.2.0
    Win32 Version: 6,0,2623,60503 @Commit: e5b72d0fb5a405d5443b571ba1d78f419bdbcdb7
[... 279 lines cut ...]
    Win32 Version: 6.0.2623.60503
    CodeBase: file:///C:/Program%20Files/dotnet/shared/Microsoft.WindowsDesktop.App/6.0.26/System.Windows.Controls.Ribbon.dll
----------------------------------------
System.Diagnostics.StackTrace
    Assembly Version: 6.0.0.0
    Win32 Version: 6.0.2623.60508
    CodeBase: file:///C:/Program%20Files/dotnet/shared/Microsoft.NETCore.App/6.0.26/System.Diagnostics.StackTrace.dll
----------------------------------------
System.Numerics.Vectors
    Assembly Version: 6.0.0.0
    Win32 Version: 6.0.2623.60508
    CodeBase: file:///C:/Program%20Files/dotnet/shared/Microsoft.NETCore.App/6.0.26/System.Numerics.Vectors.dll
----------------------------------------
System.Reflection.Metadata
    Assembly Version: 6.0.0.0
    Win32 Version: 6.0.2623.60508
    CodeBase: file:///C:/Program%20Files/dotnet/shared/Microsoft.NETCore.App/6.0.26/System.Reflection.Metadata.dll
----------------------------------------
System.Collections.Immutable
    Assembly Version: 6.0.0.0
    Win32 Version: 6.0.2623.60508
    CodeBase: file:///C:/Program%20Files/dotnet/shared/Microsoft.NETCore.App/6.0.26/System.Collections.Immutable.dll
----------------------------------------

************** JIT Debugging **************"

aliatx2017, 2024-01-11:
Thanks again nicsure! The buffer issue still occurs, even lowering it to 1. 

My prog cable (bought with my radioddity gm-30) does seem to "somewhat" passthrough audio from the radio, I also get a systray notification that quansheng dock is using the mic and I can faintly hear the TV until the buffer crash.

nicsure, 2024-01-11:
Not having that particular cable, it's difficult for me to replicate the issue to investigate. The app uses NAudio for the passthrough, which might be incompatible with that device as it seems somewhat niche. An alternative is to forget the passthrough on the app altogether and use the "Listen to this device" feature in the Windows device settings itself.

ErikS-web, 2024-01-23:
Any improvement with new version of Dock and firmware ?