nicsure/QuanshengDock #13: Digital modes with the Dock

sysadminpower2019, 2024-01-11:
Hello,

thanks for your hard work on this project.
I just wanted to ask if this Dock allows for the us of digital modes.
My understanding is that for the actual audio transmission you must create this additional cable.
So the dock is only for control of the radio itself and not for the actual audio transmission.

related to that, is there a video explaining the creation of this cable as it can still be a bit confusing for a newbie like me. I understanding that you are basically dividing the audio data at the beginning into two separate streams. Doesn't this affect the digital data coming from the usb side?
Also is there maybe some small hardware board that could be created to simplify this. Ideally if we make a hardware board then it could easily be printed and manufactured by the providers out there and might be easier for people who are new.

nicsure, 2024-01-11:
The AOIC looks like the best solution. It's an already existing project that seems to tick all the boxes. I have one coming but I can't really recommend or support it until I've been able to experiment. But from my initial research it should be able to open up all sorts of digital modes as well as simplifying the audio setup. My practical skills are sketchy at best, I'm a software guy. A video from me on construction of the cable would be disaster. I was kind of hoping another creator would do it.

ErikS-web, 2024-01-24:
See Wiki at [30-Connection-between-radio-and-PC#aioc-cable](https://github.com/nicsure/QuanshengDock/wiki/30-%E2%80%90-Connection-between-radio-and-PC#aioc-cable)

closed 2024-01-24 by ErikS-web