nicsure/QuanshengDock #15: Hamlib support for Doppler correction with Gpredict for satellite traffic ?

magellan-13016, 2024-01-13:
Hello,

First, I want to thank and congratulate you for this work. I'm F4BPP, an amateur radio involved in schools (see www.f4bpp.com) and I really love this project.

I'm a pure Linus user so, I will wait for the Linux version to try it since I've read it is your project to do a Linux version. I hope it will be a deb file or appimage format because it is the most easier way to install an application on Linux.

Like I said in the title, the most useful  function of CAT control is to correct the Doppler effect of frequencies for satellite traffic.  Since you can change the frequency trough the cable, can we hope you will make Hamlib support so we can use Gpredict to correct the doppler effect for satellite and ISS traffic ?

This would be the nicest, best and useful  feature for me and my students during space educational projects.

Thanks for reading and once again, many many thanks for this amazing project !

David (F4BPP).

nicsure, 2024-01-13:
Yes this is planned. I have little wisdom about satellite/ISS working so it's going to have to be after a lot of research and community support on this. The Linux port is on going as I type.

magellan-13016, 2024-01-13:
Wow! That's really impressive. I'm ready to give you all the help you need as far as I can. I'm currently developing software in python language that allows me to control an Icom-PCR-100 receiver via Hamlib to correct Doppler effects. The program works and all I need to do now is display the corrected frequency in real time in the main window of my software, because the frequency is actually only displayed in the Gpredict radio control module . I can provide you with the code if you're interested.

nicsure, 2024-01-13:
Hamlib is not something I've looked at yet, but it does seem like a good direction. Although I can code in almost every language on the planet, I don't really like Python, it's too different from what I'm used to and I'm not a fan of scope being reliant on indentation. I have done a few things in Python but I hated every minute of it. But that's just me, I can still read and code with it if need be. When I start looking at Hamlib your code will be very useful.

magellan-13016, 2024-01-13:
I suspected that, given the quality of your software, there was little chance of it being developed in Python, even though this language can do a lot of things and is very useful to me because it's so easy to learn and use. 

I don't know if you realize that you're going to be at the origin for the first pocket in the world (and the cheapest of all) to be computer-controlled and to correct Doppler effects! It's a world first and you can be proud of your work. As far as I'm concerned, I take my hat off to you and thank you a thousand times over for devoting your time to these developments, which are going to be so useful to the whole international amateur radio community.

na7q, 2024-01-17:
If we could get Hamlib or any standard CAT control, this would be a game changer in the community. For so many reasons.

ErikS-web, 2024-01-30:
Can be closed for now.

closed 2024-01-30 by ErikS-web