nicsure/QuanshengDock #16: Issues relating to TX

nicsure, 2024-01-13:
Almost all TX problems are going to be down to RF energy from the radio clobbering your PC, your programming cable or any related hardware. The majority are issues with the programming cable, it gets hammered by RFi and either starts sending garbage data to and from the radio or cuts out altogether leaving the software in a state where it is no longer controlling it which both cause all sorts of random events.

Get some ferrite clamps, use an external antenna, whatever you need to do to do. But I'm getting a little sick of constant issues being reported related to TX and I cannot replicate them simply because they are essentially random and hardware failure related.

ErikS-web, 2024-01-14:
Better , use an external antenna.

ErikS-web, 2024-01-21:
This is now included as [WARNING on the Wiki under Known Issues](https://github.com/nicsure/QuanshengDock/wiki/70-%E2%80%90-Tips-&-Tricks-and-Known-Issues)

and I describe suggestion to use an SMA-female to BNC-female connector/adapter

ErikS-web, 2024-01-21:
Info on [Wikipedia about radio-frequency interference (RFI) ](https://en.wikipedia.org/wiki/Electromagnetic_interference)

closed 2024-01-24 by ErikS-web