nicsure/QuanshengDock #17: AIOC cable

magellan-13016, 2024-01-14:
Hi,

I've bought the AIOC (All In One Cable) Adapter Kit For Ham Radio Baofeng & Quansheng pockets transceivers.

purchase link : [https://www.ebay.com/itm/186077832070](url)

Github repository : [https://github.com/skuep/AIOC](url)

Is this interface supported by the control software ? 

This would be a nice solution to control the transceiver without microphone and loudspeaker switched off.

David (F4BPP).

nicsure, 2024-01-14:
When I get mine I'll be able to give some advice. From all reports and what I've researched it does look like it will be compatible.

magellan-13016, 2024-01-14:
Wonderful !

ErikS-web, 2024-01-14:
Do a search on AIOC and you will find .... #8 and #6

Also... pse _**Use titles that make sense**_

Suggest Issue-starter to close this issue.

jhenkins, 2024-01-19:
Hi,

Hayden from the "Ham Radio DX" Youtube channel has successfully used this project with his AIOC cable:

https://www.youtube.com/watch?v=vgVocQpSF7s

Let's call it a "match made in heaven"! :-)

nicsure, 2024-01-19:
I also have managed to hack together a bluetooth programmer and a bluetooth audio sender. It's a bit of a mess but I can now do everything over bluetooth, meaning I can have the radio in the other room, or in the loft or wherever.

ErikS-web, 2024-01-21:
This seems to work and is now [documented in the Wiki under Connections](https://github.com/nicsure/QuanshengDock/wiki/30-%E2%80%90-Connection-between-radio-and-PC#aioc-cable)

ErikS-web, 2024-01-30:
@magellan-13016 Did you manage to get it working?

ErikS-web, 2024-02-06:
Nicsure tested himself and made an nice vidio with installation-tips.

**_AIOC Configuration_** for Quansheng Dock (date: 05-02-2024) [Link ](https://www.youtube.com/watch?v=IrnzWVdOlu8)

closed 2024-02-06 by ErikS-web