nicsure/QuanshengDock #20: Satellite Tracking

EA1BOT-Ricardo, 2024-01-15:
Possibility of satellite tracking and correction of the Doppler effect in UHF, either directly in the program or by communicating with third-party software like Orbitron or SatPC32, etc.

ErikS-web, 2024-01-15:
PSE **use titles that make sense.** 
Read it yourself on index level of issues, and you see thats there is no clear software issue. 
Its an request for ....

I think you want an "automtic doupler correction". Any idea how to do that ?

To my opinion your "request" is much to early in this development-stage.

nicsure, 2024-01-16:
I don't mind requests here, just so long as they're tagged as such and as Erik mentions, that they're not duplicates.

closed 2024-01-20 by nicsure