nicsure/QuanshengDock #22: Change Language

Conny64, 2024-01-16:
how can I change the language?

nicsure, 2024-01-16:
I only speak English. Sorry.

nicsure, 2024-01-16:
They're just what the compiler generates for multi-language support, but I haven't implemented multi-language yet. I plan to release a literals file at some point so people can make translation packs. But just being an English speaker it's not something I can do. I can use translators of course, but they're sketchy at best.

Conny64, 2024-01-17:
Thanks for the info. If you want to make it multilingual, I can help you translate it into German.

nicsure, 2024-01-17:
Thanks, I have so many things to do that it might be a while. But at that stage it will simply be a matter of taking a text file in the install, it will have lines of text in it in English, these just need to be translated to whatever language and submitted back to me. 

Again thanks for your support.

Conny64, 2024-01-17:
Don't worry, when you have finished the file just get in touch with me.

ErikS-web, 2024-01-30:
Can be closed for now

closed 2024-01-30 by ErikS-web