nicsure/QuanshengDock #23: Release Date on Download page

MikeGriffinReborn, 2024-01-19:
Can you put a release date on the main screen so we can no when this was released? It's really not clear when the release on your main page was "cut". 

I am also a .NET developer, the system we use is this. It's always moving forward numerically if you compare them and you can tell exactly when the release was "cut" from the version alone, just a suggestion  ... 

_2024.1.0118.0 = Jan 18th, 2024 (Version 1, release 0)
2024.1.0118.1 = Jan 18th, 2024 (Version 1, release 1)
2024.1.0118.2 = Jan 18th, 2024 (Version 1, release 2)_

The first three were released on the same day

Tomorrow would be ...

_2024.1.0119.0 = Jan 19th, 2024 (Version 1, release 0)_

A major release (re-write, new UI, integration, who knows)

_2024.2.0120.0 = Jan 20th, 2024 (Version 2, release 0)_

nicsure, 2024-01-19:
I'm kinda new to this whole continual development thing. When all said and done I'm only one hobby level programmer with no clue about how the real world does this stuff. I've broken MVVM in this many times etc... No doubt my procedures, code style and methodology would be ripped to shreds by a commercial development team. But yes, a "version tracker" would be a reasonable approach.

MikeGriffinReborn, 2024-01-19:
>>  I'm only one hobby level programmer with no clue about how the real world does this stuff.

Are you kidding, you're killing it, this app is ground-breaking, this whole QuanSheng "deal" is going to change the hand held market industry entirely, so we say "Thank you for your hard work" ...

ErikS-web, 2024-01-24:
Nic invented his own unique release-numbering.
And already had version numer 31.... ( see [releases](https://github.com/nicsure/QuanshengDock/releases) )

[image]

But I think it works and understandeble ;-)

@MikeGriffinReborn agree, to close this issue ?

ErikS-web, 2024-01-27:
Must be Ok ... closing Issue.

closed 2024-01-27 by ErikS-web