nicsure/QuanshengDock #24: Using radio in standalone. Scanning not working due to Remote = ON

butlerworld, 2024-01-20:
Hello,

Thanks for your hard work and what a good job.

I normally use the radio in standalone i.e. not connected to PC. The remote default is ON so each time I poser the radio on I have to go to the menau and change it which I keep forgetting to do so I don't receive anything. Could this store the last selection or default to OFF please?

Regards
Alan

ErikS-web, 2024-01-20:
Default to ON is in my opinion a better choice.

Store the selected value in eeprom would a a suggestion.

**Tip /Workaound :** buy a second radio

Second radio also better because in de QS-dock-firmware will be a static firmware and is missing functional updates that EGZUMER and Kamil (NUNU) make.

nicsure, 2024-01-20:
The reason I set that option to default to On was to prevent people complaining that the Dock wasn't working because they've set it to off and forgot to turn it back on again. If I defaulted it to Off then the majority of people trying it for the first time, it wouldn't work. The next firmware release will however address this, I can detect an attempt to read the radio by CHIRP and then automatically turn the option off.

butlerworld, 2024-01-20:
> Default to ON is in my opinion a better choice.
> 
> Store the selected value in eeprom would a a suggestion.
> 
> **Tip /Workaound :** buy a second radio
> 
> Second radio also better because in de QS-dock-firmware will be a static firmware and is missing functional updates that EGZUMER and Kamil (NUNU) make.

Unfortunately a second radio wouldn't really cure my problem as I like to use The Dock Channel Editor so would like the Dock firmware anyway.

> The next firmware release will however address this, I can detect an attempt to read the radio by CHIRP and then automatically turn the option off.

This also wouldn't cure my problem as it will still default to ON when I power cycle te radio

I like the Channel Editor in Dock as it includes the Scan list options which Chirp doesn't have

nicsure, 2024-01-20:
Solved in latest release.

closed 2024-01-20 by nicsure

butlerworld, 2024-01-21:
Hello,

Can you tell me what has changed please. I can't see any difference

Regards

Alan

nicsure, 2024-01-21:
I don't think you quite understand what the Remote option does.
It turns off the sending of UI data to the PC. It has no effect on the radio's stand alone functioning.
The ONLY effect it has is with the operation of CHIRP and now that is dealt with automatically when CHIRP is used.

So I fail to see what your problem is?

butlerworld, 2024-01-21:
Thanks for the reply.

Yes I do understand the intended function of the Remote option. What I am trying to do is report a bug or problem that I am seeing.

I will detail the problem I am having.

1. Flash the latest Dock firmware
2. Disconnect Radio from PC
3. Power cycle radio to to be sure of clean boot
4. Check the Remote option Menu 60 it is set to ON as expected
5. Select and run a scan list i.e set scanner to scan through a list of memories (16 memories in my case)
6. The squelch is never opened when I expect a sigmal to be received so scanning never stops
7. Turn OFF Remote option Menu 60
8. Select and run a scan list i.e set scanner to scan through a list of memories (16 memories in my case)
9. The squelch opens when a signal is received and scanning stops while that signal is active

So I have to remember when I power on the radio to set the Remote option OFF or I will not recive anything

I live close to an RAf airbase and a 2m repeter so I get quite a lot of signals througout the day.

I hope this ,helps and thanks for the great firmware and software

Regards

Alan