nicsure/QuanshengDock #25: BlueTooth interfacing - how2

ErikS-web, 2024-01-20:
I read in #17 that you are experperimenting with that BT-interface next to AIOC-interface.
Very interesting, because HF interference will me less because it is wireless.
Compared to your initial hardware-wiring (with or without my suggested red-wire to microphone ;-)

I had a look on AliExpress and saw 2 types

- **BL-1**

[image]

- **BPC-1** 

[image]

Both cost in the range 15... 22 euro.

What type are you using ?

nicsure, 2024-01-20:
It's the same as the first image, the same one Andreas demonstrated in one of his videos. I opened it and tapped the points for the audio in, audio out, power and ground. The audio in and out lines are not used by the device, but the solder points are still there, just leading nowhere. Then these 4 wires are fed to a bluetooth audio sender. Although the power voltage is not the same as the audio sender uses it still works, I suspect it has some voltage regulation inside it. Then just pair both devices and you're good.

nicsure, 2024-01-20:
I'll make a walkthrough with photos, but my implementation is quite messy. I'll have to do my best to clean it up.

ErikS-web, 2024-01-20:
So you use it as a computer interface for the controls and not as a sound card.
And the Audio leads you to the Line-In/Out of your PC

3.5mm middle pin (Mic) to Line-out of PC
2.5mm Top-pin (LS) via 47 K-Ohm to Line-in or PC
2.5mm Connector side to Gnd

Is the 3.5mm Top-pin not connected to power the device?

I'm curious about the photos

nicsure, 2024-01-20:
No, I opened the bluetooth programmer and tapped off the points for audio in, audio out, power and ground from where the jacks are soldered to the board. This gives me 4 wires coming out of the programmer. These are then wired into two 3.5mm jacks and a micro usb plug which then plug into the Bluetooth audio sender. From the PC I just pair to both the programmer and the bluetooth sender.

nicsure, 2024-01-20:
https://www.ebay.co.uk/itm/284907182377?hash=item4255ca9929:g:IX8AAOSwIHxlCve4
https://www.ebay.co.uk/itm/235380965551?hash=item36cdcc80af:g:CP0AAOSwlhNlngpg

ErikS-web, 2024-01-20:
To my opinion the first one. Can do RX or TX. Mostly not at the same time.

Second one is the BL-1. 

Waiting for your wirings photo's ;-)

nicsure, 2024-01-20:
Yeah it can do mono at the same time, not stereo.

ErikS-web, 2024-01-21:
Mono ? Its not in the specifications. What is the trick ?

ErikS-web, 2024-01-22:
[image]

<br> 

How do you connect that flat **Model: RX-TX-10** for the audio transmissions of Ls and Mic.

How do you _force it to connect Duplex_ (**RX and TX)** and in **Mono** ?

ErikS-web, 2024-03-12:
I will close this issue. I will not continue this as devellopment
We can always re-open or start new Issue.

p.s. I now use a wireless (BT) headphone on the PC

closed 2024-03-12 by ErikS-web