nicsure/QuanshengDock #27: Use WIKI for documentation of QS-Dock  [request]

ErikS-web, 2024-01-20:
Now there is so much functionality in Dock....

Suggest to activate the Wiki as Online documentation.

I can provide `extra hands`

nicsure, 2024-01-20:
I enabled the Wiki in the settings. Not sure what to do now though.

ludwich66, 2024-01-20:
or see  https://github.com/ludwich66/Quansheng_UV-K5_Wiki/wiki/Quansheng%E2%80%90Dock
Is this Part OK of you? 
Plaese check Schematik.

Greatings ludwich

nicsure, 2024-01-20:
Looks good. You should be able to edit this wiki as I've enabled public editing. I'll figure out how to add you as collab if you want?

ErikS-web, 2024-01-20:
Ok, had some brainstorming, and made an index as initial setup.
Will continue in next days. As I am also working on Kamilsss Wiki

Feel free to help. Lets communicatie via Telegram.

ErikS-web, 2024-01-20:
@ludwich66  / Ludwich

You could also include my "Alternative to Hardware Configuration" see picture in #12.

ErikS-web, 2024-01-21:
[Wiki is under construction ](https://github.com/nicsure/QuanshengDock/wiki)

ErikS-web, 2024-01-21:
This can be closed

closed 2024-01-21 by ErikS-web