nicsure/QuanshengDock #28: Radio firmware CRC CHECK FAILED! FIRMWARE NOT VALID! in UVTOOLS-link [BUG]

ErikS-web, 2024-01-20:
In README.MS section
https://github.com/nicsure/quansheng-dock-fw?tab=readme-ov-file#requires-quansheng-dock-v0313q

erro when trying to flash via UVTOOLS

_Loading file from url: https://github.com/nicsure/quansheng-dock-fw/releases/tag/0.31.1q/firmware.packed.bin

WARNING: **CRC CHECK FAILED! FIRMWARE NOT VALID!**
Make sure to choose a flashable bin file. 
Error while loading firmware, check log above or developer console for details._

Link should be

https://egzumer.github.io/uvtools/?firmwareURL=https://github.com/nicsure/quansheng-dock-fw/releases/download/0.31.1q/firmware.packed.bin

nicsure, 2024-01-20:
I just got the URL wrong in the link, is fixed now.

ErikS-web, 2024-01-20:
Suggest to add text [🗲FLASH WITH A BROWSER🗲](https://egzumer.github.io/uvtools/?firmwareURL=https://github.com/nicsure/quansheng-dock-fw/releases/download/0.31.1q/firmware.packed.bin) in the README.md

ErikS-web, 2024-01-20:
OK, fixed

closed 2024-01-20 by ErikS-web