nicsure/QuanshengDock #30: Youtube URL in bottom line, does not start YT (in Settings) [BUG]

ErikS-web, 2024-01-21:
In Settings there is a YT URL on the bottom line.
When I press it, **no window opens** to view your _beautiful recordings._

[image]

Workaround.... [See the Youtube links on the Wiki](https://github.com/nicsure/QuanshengDock/wiki/80-%E2%80%90-Reviews-on-Youtube)

nicsure, 2024-01-21:
No it copies to the clipboard. Opening a browser window requires starting a process and that can upset virus scanners.

ErikS-web, 2024-01-21:
I'll put this info in the Wiki

ErikS-web, 2024-01-21:
Suggest to update this url in next release _**to the Wiki**_

https://github.com/nicsure/QuanshengDock/wiki

ErikS-web, 2024-01-21:
Suggest to update this url in next release to point to the Wiki

https://github.com/nicsure/QuanshengDock/wiki