nicsure/QuanshengDock #32: Cant Start the new 0.31.3

Blinky122, 2024-01-22:
Hello,
when i will start the new version. i become this issue

[image]

ErikS-web, 2024-01-22:
What operating system and version ?

Did you use the latest release [0.31.3q](https://github.com/nicsure/QuanshengDock/releases/tag/0.31.3q) ?

nicsure, 2024-01-22:
There are some config changes, It looks like I didn't handle some conversion from the older format. Clearing the app's config might work. Config is stored in the User Documents folder.

ErikS-web, 2024-01-22:
Suggest to update to latest release of [QuanshengDock](https://github.com/nicsure/QuanshengDock/releases)

and the [latest Firmware for radio]( https://github.com/nicsure/quansheng-dock-fw/releases)

closed 2024-01-25 by ErikS-web