nicsure/QuanshengDock #35: Dots and Comma`s ? How2 enter frequency

ErikS-web, 2024-01-25:
Initial screen

[image]

Switch to VFO
 
[image]

XVFO
Starts with Comma in frequency

[image]

 
With DOCK-keyboard&Mouse type 145.000
 
[image]

ENTER
GIVES 1300,00000
 
[image]

145 Enter gives

 
[image]

145.4 on **Numeric part of keyboard**
 
[image]

ENTER
Gives

[image]

Actual Dock and firmware. See printscreens
Windows 10, with US-Int keyboard and NLD-settings of windows

nicsure, 2024-01-25:
Must be something to do with language settings in windows, all mine are just dots I don't get commas.

nicsure, 2024-01-25:
It looks to be ignoring the "." and thinks you're entering 1454 so it's clamping it to 1300. What happens if you enter "145,4" with a comma ?

nicsure, 2024-01-25:
Also try clicking on the frequency and then editing it. From what I can tell it has something to do with the way C# converts text to numbers and it is different depending on how Windows is setup language wise.

nicsure, 2024-01-25:
https://github.com/nicsure/QuanshengDock/releases/tag/0.31.8d

See how that works. It's hard for me to test as I don't have my PC setup for that language mode.

ErikS-web, 2024-01-26:
Dot and Comma’s part-2

Intital screen with DOT
[image]

Selecting XFO ...  XFO screen with Comma
[image]
 
And strange stations in memory schannels (mentioned/had this problem earlier)
Turning VFO give error: Out of Range
 
[image]

Restarting Application
and press **Spect**
 
[image]

Left DOT and Right Comma ????
Press **Analyser**
Now all Comma’s
 
[image]

The Numeric-keyboard “.” Is not accepted.
I have to use the Comma (On ASCII-keyboard To the right of M-key)
The “.”(dot) net to “,”(comma) is also not accepted

 
[image]

ErikS-web, 2024-01-26:
Communication with the radio is also disrupted after a crash.
Then I have to power the radio off and on.

Is there some control on radio communication, such as a check-sum? That could make communication with the radio more stable.

nicsure, 2024-01-26:
You've got some corrupted config data, you have ??? for the mode and 25 for the bandwidth, something is seriously messed up in the app's saved config

ErikS-web, 2024-02-02:
Suggest to test the latest release [0.32.6q](https://github.com/nicsure/QuanshengDock/releases/tag/0.32.6q)

I am happy ;-)

ErikS-web, 2024-02-02:
[image]

I finally can close this ISSUE. 

**_Thanks for solving it_**. 👍 01

closed 2024-02-02 by ErikS-web