nicsure/QuanshengDock #38: Remote Audio via RDP no function

DL9BO, 2024-01-29:
Is there a possibility to use the RemoteAudio drivers when using Remote Desktop?
My tests were not successful.
Or is there a workaround?

Best regards
DL9BO Bernd

nicsure, 2024-01-29:
When all said and done audio is just simply input and output devices. It might be better to disable passthrough in the app completely and handle remote audio over RDP independently.

DL9BO, 2024-01-29:
Thanks for the quick feedback.
I don't use "passthrough" anyway, because error messages occur regularly. As soon as an RDP connection is established, I switch to the remote audio drivers for input and output. However, there is no sound at all via the connection. 
I have carried out extensive rdp group policy settings.
Have you tested it yourself?
I think some users here are also playing with the idea of implementing remote access. Therefore, if you can find a little time, it would be nice if you tried it out yourself.

nicsure, 2024-01-29:
Well of course there's no sound. Can you hear the sound being picked up from a microphone plugged into the PC you're RDPing into? No, of course you can't. It's exactly the same thing. To the PC the radio's sound is no different to a microphone plugged into it. So that's how you need to approach it.

closed 2024-01-29 by nicsure