nicsure/QuanshengDock #39: Out of bounds exception

timothycolley, 2024-01-29:
A relativly easy fix for you here,

There is an out-of-bounds exception on the XVFO memory list controls.

it occurs when the list is blank and you attempt to use the spinner control. 

this behaviour caused the app to become unresponsive.

Screenshot attached.

[image]

nicsure, 2024-01-29:
Nice catch. Will address it asap.

timothyjcolley, 2024-01-30:
I also tested the android app with the latest fw. No response to or from the radio. I tries it using the prog cable va a usb-c otg adapter on a Samsung galaxy flip 5.

I don't believe that the serial port is showing up under android 14. This is either an sdk related issue (maybe you need to auth the app as an administrator of the device, or it can't be done without root. But these are guesses.)

Can I assume that you figured that this would be an issue as you did an additional side build of the same unity program and hoped for the best?

nicsure, 2024-01-30:
Currently, UI is a bit of a fishing trip. I knew Android was going to be tricky due to so many different security models and access restrictions over different Android versions. Unity offers little in terms of defining privs, it's all done automatically.

nicsure, 2024-01-30:
It looks like Android is going to have to go on a back burner. It seems that the System.IO.Ports is not supported in Android. It requires some stupid aar package to be compiled along side in order to access ports. There is a plugin made by someone trying to charge 80 bucks for it, but I'll be damned if I'm paying $80 for something I'm doing for free. Android users are going to have to accept that their device is shit, their OS is shot and thus they can do one.

nicsure, 2024-01-30:
Sorry I deleted the comments about the other version as I've now withdrawn it.

nicsure, 2024-01-30:
Should be fixed next release

timothyjcolley, 2024-01-31:
Yeah, to be honest, all the system level stuff has been sandboxed on
android in recent years. Anything useful is inaccessible now, unless you
root, and that disables pay apps or voids your warranty.

Its not a shit OS, it just isn't supposed to be as extensible as a full
desktop class os.

Tbh, I love your main app, and I use it on a Microsoft surface tablet, that
way I get the feel of a real interface, without having to use android for a
touchscreen.

I would suggest recommending using a touch capable laptop or windows tablet
for your app.

Also, see if you can find a compatible programming cable with an an
official prolific chip in it. If you find and recommend one, it removes a
point of failure from the overall process and makes for happy users.

I'll continue testing for you and report any other bugs or suggestions via
github.

Many thanks

Tim

nicsure, 2024-01-31:
Yeah I had the fake prolific problem with my Baofeng cable too. Took a few minutes to Google and locate the older driver package that works. At the end of the day, the guts of the programmer is a USB to TTL serial chip, the CH340 or FDTI variants should work just as well I imagine. But even those I believe have fakes kicking around. My "Frankenstein's Monster" setup I'm putting together as an alternative to the AIOC uses a CH340.

ErikS-web, 2024-02-02:
Suggest to test the latest release [0.32.6q](https://github.com/nicsure/QuanshengDock/releases/tag/0.32.6q)

timothyjcolley, 2024-02-03:
The interface is touch capable. Windows uses mouse emulation for this. You did a great job of making the interface with big friendly finger size buttons. So it already great.

ErikS-web, 2024-02-03:
Does this mean Issue is solved ?

If yes, please PRESS `Close with comment` this Issue. _See on the right below_

ErikS-web, 2024-02-04:
@timothycolley pse respond

closed 2024-02-04 by nicsure

timothyjcolley, 2024-02-05:
Sorry guys, didn't see that last post till just now (full time dad here, lol)

Thnx for closing it.