nicsure/QuanshengDock #40: Compatibility with Winlink / Packet / VaraFM

sysadminpower2019, 2024-01-30:
Hello,

does any part of this project (the docker and / or cable setup) provide for some digital / packet work.
It would be really great if all of this would integrate into the digital side and allow for things like winlink or even aprs.

nicsure, 2024-01-30:
It is planned yes. I need two radios to continue in that direction, I have another coming soon. SSTV, Messaging, and KISS mode TNC operation are planned.

sysadminpower2019, 2024-01-30:
Thanks for the reply.
What would be the hardware requirements? Just the special homemade Dock
cable?

> It is planned yes. I need two radios to continue in that direction, I
have another coming soon. SSTV, Messaging, and KISS mode TNC operation are
planned.
https://ci5.googleusercontent.com/proxy/jx3vVovZsRpEb0aA6GtUhBan86PSVOnebkJ_DkH_l5xqqz1sBYHcsAS2mUxMdSOiYjpKwBCqvYL6EfozTIXnzv2CSxIheOLx1kOKxsWxxH4HbTEAe08ID5_JeuNKZEqxmJBAypmbsmyWnZQzXUMnJumSkVESxnjEWWdiBtXLMgpSzKRZ7Zx9MvSTzborXH6bcp3z_DuhJGvJ9CMT5fbyGTFM4mabWcY8vOwADQMXVHGqhvWivg=s0-d-e1-ft#https://github.com/notifications/beacon/AJWXABTSO2DD7Q6BNE2MZVTYRBE27A5CNFSM6AAAAABCQL6QYWWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTTSGJFBY.gif>Message
ID: ***@***.***>

nicsure, 2024-01-30:
I had to edit your comment and remove the last part as it could have compromised your email security.

Something like the AOIC would probably be required.

ErikS-web, 2024-01-30:
Suggest to READ the [Online Manual/Wiki](https://github.com/nicsure/QuanshengDock/wiki)

Than you have an idea what is working.

And use _Issue_ wnhen it is not.

ErikS-web, 2024-01-30:
Why the question ?

Would you like to _participate in development ?_

ErikS-web, 2024-01-31:
Closing this Issue. 

I am also curious how QS dock will develop.

There are quite a few important technical issues that require attention. Supporting multiple platforms but also something simple that turns out to be difficult "periods and commas.

But this is all a hobby and free time and mainly own interests. But feel free to help and assist or [support.](https://github.com/nicsure/QuanshengDock#support)

closed 2024-01-31 by ErikS-web

nicsure, 2024-01-31:
Unfortunately I can't control the behaviour of .NET or Windows. If the underlying system decides to use a comma instead of the obvious dot because of culture/language settings, it's going to cause problems. I have modified all floating point parsing functions to try both dot and commas when converting from strings. If that's not working then I don't see what else I can do.

sysadminpower2019, 2024-02-03:
yea i'd love to help but i am a sys admin not a developer

nicsure, 2024-02-03:
And I wash dishes for a living.