nicsure/QuanshengDock #44: Codebase in GIT

SigmazGFX, 2024-02-08:
I'm just curious if the release routine is still wonky because the releases that arrive to the public Git repo don't include the latest commits other than the readme updates and the new compiled zip which was not there in the  0.32.4q release.

It's fine if you are keeping it closed. I was just trying to follow along in the code and the later release features are not available to review.

Thanks and 73

djex, 2024-02-09:
I noticed this as well. Also the source code zip in the latest release 0.32.8x is actually the source for version 0.32.6q but the compiled exe is 0.32.8x. I feel this may be a mistake or bug as the latest firmware source for 0.32.8x seems to be the right released version.

I too was following along as I am quite interested to see how communications are done between the radio and PC over serial. Thinking of getting a few of these radios to play around with.

nicsure, 2024-02-11:
Yes, there's something wrong with my Visual Studio's source control. Every time I save it synchronizes to a private repository so I don't end up with hundreds of commits on the main branch. Then it's supposed to update the main branch on a release build. I'll have to update it manually somehow.

I use VS-Code for the firmware and the source control on that seems to be fine. That's why the firmware is correct to source.

nicsure, 2024-02-11:
Oh boy this is going to be a BIGGGG commit.

djex, 2024-02-12:
Yeah it looks like the source has not been updated since v0.27.6q. The source zip for your latest version v0.32.11q when compiled is actually version v0.27.6q. 

What do you use for your private repo? I use gitea myself and it works quite well for a local repo.

nicsure, 2024-02-12:
I just use the one built Into Visual Studio and it's supposed to handle it all for me, including the sync to GitHub. I'm gonna attempt a manual commit now, hopefully it won't all go tits up.

djex, 2024-02-13:
That seemed to do the trick. Compiling master now gives the latest release version. Thanks for the manual sync. Unfortunately I'm not too familiar with how the built in Visual Studio git works. I have a local git server using Gitea and I just push to that from Visual Studio and do everything else I need in Gitea.

nicsure, 2024-02-13:
Lol, it stopped working because VS had signed me out of my Microsoft account for some reason. Some 2FA BS I hadn't bothered with. Should be okay from now on hopefully.

closed 2024-03-04 by nicsure