nicsure/QuanshengDock #46: [FEATURE] Add received channel info on screen in XVFO-scan

ErikS-web, 2024-02-11:
QS-Dock version 0.32.8x, XVFO, SCAN

Now the text is "Scanning list ..." or Channel info when mouse on selected channel,

**Situation** 
During scan a station is detected and this is drawed with colour. 

**Feature-request**
Is it posible to add the Frequency and Channelname on display as long as monitor is receiving om that frequency ?

And what can be **_follwo-up actions_**....
SPACE-bar: stop scanning and stay on this frequency
Right arrow: Continue scanning directly
Left arrow: Reverse scanning (if possible)

nicsure, 2024-02-11:
The frequency and channel name is displayed on the main XVFO window, do you mean duplicate that in the scan window?

ErikS-web, 2024-02-11:
YEs but ONLY when signal is received.

nicsure, 2024-02-11:
So, for example. Where "Scanning List Whatever.." is displayed in the scan window, when monitor is engaged (either manually or automatically) this changes to something like "Monitoring <channel name> on <channel frequency>"

ErikS-web, 2024-02-11:
[image]

p.s. of course not the white background.

nicsure, 2024-02-11:
Implemented

ErikS-web, 2024-02-12:
Beautifully realized in new release [0.32.11q](https://github.com/nicsure/QuanshengDock/releases/tag/0.32.11q)
 
[image]

Thank you 🥇

closed 2024-02-12 by ErikS-web