nicsure/QuanshengDock #51: [WISHLIST] PLEASE CREATE SOMETHING LIKE FOR R3TS I BEG

kennykyle, 2024-02-15:
I have a physical disability and i can't interact with physical radios but i can use a phone or a computer fully. I really want to learn radio and HAM but it's almost impossible without some extra help. COULD YOU PRETTY PLEASE make something like this for the retevis r3ts? Ill pay you XD

ErikS-web, 2024-02-15:
Why the `Retevis rt3s` ? 

The Retevis rt3s is a **_DMR radio_** and the UVK5 is **not** !

All the knowledge and resources of the UVK5 radio are known and developed. 
Both the open source firmware on the radio and the QS-Dock on the PC.

I think you are unfortunately asking about the (still) unknown path.

nicsure, 2024-02-15:
I've agreed to look into it to make a basic remote system for that radio as it has open-source firmware.
It won't be a version of QD but a separate project altogether. You can move this from the active list if you like.

ErikS-web, 2024-02-16:
Yep it's a SPECIAL. 

And will get a follwup at a different Github place.

Suc6

closed 2024-02-16 by ErikS-web