nicsure/QuanshengDock #52: [Feature] ideas for XVFO Keyboard key-assignment

ErikS-web, 2024-02-18:
Suggest 

`F` set Mode FM.  `f` for FM- narrow and `F` for FM-wide
`M` of `m` for max Mic Gain (100%)
`P` for vary Power as + key
`Q` of `q` for toggle sQuelch open
`T`or `t` for transmit 1750 Khz

nicsure, 2024-02-18:
I can set F to toggle between FM Wide and FM Narrow, but F is just F there's no distinction between upper and lower case.
M is already used to toggle through modes, suggest another key for that.
Power on P will be in next release.
Squelch on Q will be in next release.
T I'm not sure about, do you mean transmit a 1750Hz tone?

ErikS-web, 2024-02-18:
`F` as a quick key to go2 FM-wide and than toggle Wide/Narrow

`M` is indeed already in use and I mentioned it on the [key-assignment on the Wiki](https://github.com/nicsure/QuanshengDock/wiki/50-%E2%80%90-Specific-Functions-explained#keyboard-key-assignment)
Perhabs use `G`ain or `V`oice

P and Q nice 

`T` indeed to perform the 1750 Hz tone

nicsure, 2024-02-18:
In next release :
F = FM Wide/Narrow toggle.
G = Mic gain
S = Toggle squelch mode, (same function as clicking the button)

Need to work on T bear with me.

ErikS-web, 2024-02-18:
Suggest ... just like DTMF active and pressing (numeric) keys..

ErikS-web, 2024-02-18:
Next release will have extra keys ... already mentioned in the [Online manual](https://github.com/nicsure/QuanshengDock/wiki/50-%E2%80%90-Specific-Functions-explained#next-release-will-have)

ErikS-web, 2024-02-18:
Q or S for sQuelch ?

nicsure, 2024-02-19:
Q is the same as clicking Open to open the squelch.
S is the same as clicking Squelch to change the squelch setting.

nicsure, 2024-02-19:
All done, managed to get T to send a 1 second 1750Hz tone when pressed during transmit.

ErikS-web, 2024-02-19:
Nice, but why can GAIN not set to 100% (Is now from 0 .. 96%)

Suggest to toggle to MAX (100%) and NOT rotate back from 0

nicsure, 2024-02-19:
Well I'm sorry I can't do everything exactly as you suggest, the difference between 100% and 96% is not even humanly noticeable.

ErikS-web, 2024-02-19:
Well,  I think to hear it is somewhat stronger audio signal at 100%.
And do not have any reason to start from 0.

At all my radio's I have to set Menu.31 `Mic` to MAX value +15.1dB to be able to have good volume with external microphone.

ErikS-web, 2024-02-19:
Editing the Wike now with latest release.

How2 explain SQ-1... -4 ? 
So I user SQ-R in combination with "Bar". Or how do you call that ?

Selecting these item don't show anything changing on my screen

ErikS-web, 2024-02-19:
See the temporary label: 
https://github.com/nicsure/QuanshengDock/wiki/50-%E2%80%90-Specific-Functions-explained#next-release-will-have-03216q

ErikS-web, 2024-02-19:
Above mentioned Temporary label removed.

All keys now here [keyboard-key-assignment](https://github.com/nicsure/QuanshengDock/wiki/50-%E2%80%90-Specific-Functions-explained/_edit#keyboard-key-assignment).

Could you verify ?

nicsure, 2024-03-01:
Keys look good.

ErikS-web, 2024-03-02:
Thanks to user **GikaH** for additional feedback

closed 2024-03-02 by ErikS-web