nicsure/QuanshengDock #53: [Feature] XVFO: vary Mic-gain and Power during transmit

ErikS-web, 2024-02-18:
In actual version it is not possible to ..

- vary MIC-gain during transmit
and
- vary transmitted Power during transmit

combined with key-assigments as sugegsted in #52 

I am not sure if it is possible during TX, but lets find out.

nicsure, 2024-02-18:
Power is adjusted by the bias on the PA. It would not be a good idea to change that while the PA is active and I suspect the chipset would be hard-wired to prevent it. Also it would require serial communication to the radio during transmit phase, which is something I try to avoid at all costs. It would also involve dealing with multiple inputs, I'd need to check for keys being pressed while other keys are pressed and/or mouse buttons etc.. Something I'd rather not do.

ErikS-web, 2024-02-18:
I also suspected that it wouldn't be possible (easily).
Then we DON'T do it either

closed 2024-02-18 by ErikS-web

EarwigBrian, 2024-02-18:
All done now then.... Great work cheers NIc Sure enjoyed the trip hope you have more successes with all you develop in the future ;) 73