nicsure/QuanshengDock #55: [Feature] XFVO scanner: Logging received frequency

ErikS-web, 2024-02-18:
In XVFO-scan add a button `LOG`, that activates logging of received signals

When a station is received. XVFO-scan will (create and ) write to a logfile 

and logs _all technical details that we can think of ._... Moment (date and time), Frequency, Mode, Signal strenght, Duration, (optional channel name)

Filename logfile includes starting moment of scan.

nicsure, 2024-02-18:
That shouldn't be too hard to do.

ErikS-web, 2024-02-18:
If you then run an (separate) audio recording on the PC. You can see exactly what time there is on a certain frequency, perhaps something interesting can be heard.

nicsure, 2024-02-19:
Done

ErikS-web, 2024-02-19:
Is this included in 32.16.q ?
I see a directory d:\Documents\QuanshengDock\**scanlogs**

But now logging.
How2 activate and disable ?

ErikS-web, 2024-02-20:
Manually modified Parameter `ScanLogger` from _False_ into _True_ in file _**app.config**_

Now the scanner Logs 👍 

[image]

ErikS-web, 2024-02-20:
Oeps... If I only had the better look at SETTING.... and now I see 
[image]

New picture for yhe Wiki ;-)

ErikS-web, 2024-02-20:
See Wiki

[logging-received-frequency-in-csv-format](https://github.com/nicsure/QuanshengDock/wiki/50-%E2%80%90-Specific-Functions-explained#logging-received-frequency-in-csv-format)