nicsure/QuanshengDock #59: [Feature] QDNH - Overview of current Settings

ErikS-web, 2024-02-25:
When program is started an "Overview of the settings is given".

Then I changed the settings and I would like to see this overview again.

There is no reaction on ENTER other than "Enter Command (? for help) #

So I suggest use Enter to give Overview.

p.s. Unclear how to use the N command. Because 2 default ports are defined.

nicsure, 2024-02-25:
You can get a readout of a particular setting by just entering the letter by itself, but yes, a full overview on "enter" would be sensible. 

The ports used are sequential, so if you enter "N 12345" you'd set ports 12345 & 12346. Basically you just enter the first port and the second one is always +1

ErikS-web, 2024-02-25:
Ok, that works,
And 2-ports is now explained on Wiki

As I am lazy, I would also like to have the **Overview with ENTER** and the **local IP-address** in the overview ;-)

nicsure, 2024-02-25:
I just updated with that implemented.

ErikS-web, 2024-02-28:
Yes very nice. It reacts on ENTER with overview 👍 

And Line **Local Hostname** is a good suggestion.
_This is a convenient way, instead of looking up IP numbers._

Used it on a (slow) testing laptop with QD and that work.
Mus say that using the exact ip-numbers was detecting more quickly. But thats Windows ;-)

*********************************************

My screen ....

Quansheng Dock Network Host 0.01.04q

(* shows currently selected)

Available input devices
  0 - Disabled
 *1 - Lijningang (High Definition Audio Device)

Available output devices
  0 - Disabled
  1 - Echo-onderdrukkende telefoon met luidspreker (CONEXANT USB AUDIO)
 *2 - HP E243m (Intel(R) Display Audio)
  3 - Luidsprekers (High Definition Audio Device)

Available serial ports
  0 - Disabled
 *1 - COM7
  2 - COM11

Network Ports           18822, 18823

**Local Hostname          DESKTOP**

Login Password          NOT SET

Audio Latency           100

Enter Command (? for help) #

closed 2024-02-28 by ErikS-web