nicsure/QuanshengDock #60: [FEATURE] XVFO: add VOX-button

ErikS-web, 2024-02-28:
Would like to have `VOX`-button at second XVFO-screen

nicsure, 2024-02-29:
I was contemplating this, but it's kind of tricky. I'll see what I can come up with.

nicsure, 2024-03-03:
This is not simple, you have VOX sensitivity, but you also don't want it going into transmit when you put your coffee down near the microphone and you don't want it keying down too quickly between pauses in speaking. That's 3 extra settings to deal with. Also I think it's prudent to automatically disable VOX on change of frequency.

nicsure, 2024-03-04:
17q has VOX now, sensitivity can be adjusted in the settings. Night need a little tweaking though.

ErikS-web, 2024-03-04:
it works, after some tuning 👍 

[image]
[image]

closed 2024-03-04 by ErikS-web