nicsure/QuanshengDock #61: [BUG] XVFO Scanner - Blue lines,  after receiving a station

ErikS-web, 2024-03-02:
QD 0.32.16q XVFO scanner mode,  using a scan **RANGE 2 meter** (144-160 Mhz)

When a channel is received, a purple line is drawn. AFTER 4 seconds, or as soon as the signal disappears, the scanner continues.
But, the trigger then draws a **_blue line on all channels afterwards_**, with the same signal strength, **_while nothing is received._**

[image]

nicsure, 2024-03-02:
Yes I know about that one, it's because the whole range is treated as a single channel, so the signal history is being applied to all the frequencies in the range. I think probably removing the signal history bars for a range would be the easiest option.

ErikS-web, 2024-03-03:
Yes, let's try to come up with a functional solution somewhere.

Because I do like the screen, also in mixed mode. Range(s) and specific frequencies 👍 

In this picture: Freq, Freq, Range, Freq

[image]

nicsure, 2024-03-04:
This should be resolved next release.

ErikS-web, 2024-03-04:
Published SOURCE code under rel 18 is version 17.

[image]

nicsure, 2024-03-04:
Yeah I forgot to sync before I compiled the release. The main branch is .18q though

ErikS-web, 2024-03-04:
But look here... nice picture .... 👍 

[image]

ErikS-web, 2024-03-04:
Yep... i compiled loaclly and it was 17 ???? That is how I detected.

Aha I now also see what is mentioned in release notes
[image]

[image]

ErikS-web, 2024-03-04:
Perhabs delete release 18 and insert it again on Github ?..

This issue can be closed. Scanning RANGE is now OK.

closed 2024-03-04 by ErikS-web