nicsure/QuanshengDock #64: [BUG] VFO

ErikS-web, 2024-03-06:
Release 0.32.20q (and also source 21)

When VOX is enabled

VOX on goes TX for 1 second RX for 1 second and repeats. Even if the GAIN is LOW

nicsure, 2024-03-07:
Not for me it doesn't. Please supply all settings you have related to VOX and microphone levels.

ErikS-web, 2024-03-07:
It's tricky. Had a XM5 mic+headphone on and connected via BT to PC.

Sensitivity at approx 28.7.
Will do some test next week.

nicsure, 2024-03-12:
I've been testing this and as long as the microphone level is correctly applied, it works fine for me.

closed 2024-03-13 by ErikS-web

EarwigBrian, 2024-03-14:
VOX does not work it transmits as soon as you hit buttton even with mic turned completely down in XVFO but it works ok on radio Sent from Mail for Windows From: ErikS-webSent: 13 March 2024 19:57To: nicsure/QuanshengDockCc: SubscribedSubject: Re: [nicsure/QuanshengDock] [BUG] XVFO - VOX goes ON and OFF (Issue #64) Can not reproduce the error. Closing issue

EarwigBrian, 2024-03-14:
Ahhh lower than 28 then got it thanks

EarwigBrian, 2024-03-14:
And tested works fine and my audio was good had report from G0WCO on GB3NR testing VOX my setting was 24.10 works well logitec web cam for mic.