nicsure/QuanshengDock #66: [BUG] exit XVFO, always have to wait 5 seconds before PTT work on first screen

ErikS-web, 2024-03-06:
Rel 0.32.20q

When exit XVFO screen and go back to First screen, the PTT does not work directly.
I have to wait approx. 5 seconds before it does function.

I think this is also the case when starting QD. (But than I have to mouse mouse to Unlock, what takes some time ;-)

nicsure, 2024-03-07:
it's intentional

ErikS-web, 2024-03-07:
Some synchronisation ?
Or else for what reason ?

Would like to mention and explain this (behaviour) on the [Online Manual / Wiki](https://github.com/nicsure/QuanshengDock/wiki)

nicsure, 2024-03-08:
It's a cool down, keys can be "stuck" on in Windows sometimes and it would be bad if the application started and went in TX straight away. There's also a delay before the main window opens due to app initialization, people can sometimes start pressing keys before the main window opens. When the app starts or switches between XVFO and Normal mode, there's quite of a lot of communication between the radio and the PC to get things initialized. I need to make sure all of that is completed before allowing TX. Besides, I don't think anyone should be transmitting within five seconds of startup, you're supposed to wait at least that long to make sure the frequency is clear.

nicsure, 2024-03-12:
As this is not a bug and not possible to resolve I'm closing it.

closed 2024-03-12 by nicsure

ErikS-web, 2024-03-12:
Mentioned in the [Wiki](https://github.com/nicsure/QuanshengDock/wiki/50-%E2%80%90-Specific-Functions-explained#exit---wait-5-seconds)