nicsure/QuanshengDock #75: F-5 no longer brings up Spectrum Analyzer after going to Egzumer and back to your driver

MikeGriffinReborn, 2024-04-02:
I like your driver much better, the scanning specially among other things, however, I cannot get F-5 to work, if I re-install egzumer it works, strange? I can live without the Spectrum Analyzer but it would be nice to have. I have two Quansheng UV-K6's and they both now exhibit this behavior

MikeGriffinReborn, 2024-04-02:
Just a heads up when I press F5 my channel goes from say "M13" to "M5--" 

As if I trying to program the channel number or something ... Probably something stupid I did

closed 2024-04-02 by ErikS-web

MikeGriffinReborn, 2024-04-04:
Okay, no worries, I have yours installed. I have the Q-Dock app on my laptop, works great. I don't have the true cable for TX yet but that will come, hoping those end up getting mass produced for purchase somewhere if not will make one. Thank you.

EarwigBrian, 2024-04-04:
Mike Griffin From a avid fan and daily user of this software. I advise the AIOC connection I bought one. Previously I did make cable it worked but using hand mic was a bit of a chore. So now with AIOC, I press can space bar on PC and am TX's good audio reports, scanning 2m 70cms Air and marine bands.