nicsure/QuanshengDock #84: QUERY: Serial Port Protocol Documentation

KDMcMullan, 2024-07-17:
I have a very specific application for the Quansheng which will require me to control the frequency, bandwidth, CTCSS and output power level of two of the HTs. I believe these features are exposed by the QD firmware. (As a nice-to-have) I'd like to monitor battery voltage.)

Is the serial port protocol documented anywhere, please?

samoyedfranek, 2024-11-19:
have you done everthing at your own?

KDMcMullan, 2024-11-21:
I've not done anything. I'd like to. QuanshengDock seems to expose the features I need, as a sort of CAT, but I have no idea how to find the protocol.

samoyedfranek, 2024-11-21:
I only find a way to read serial info

BranoSundancer, 2024-11-21:
The CAT implementation (Yaesu FT-991 partial emulation) is done in QuanshengDock XVFO mode. The serial interface does not contain any CAT protocol and perhaps never will, because there is not enough memory in the radio to store such software piece. For resources about QD XVFO CAT implementation please see [discussion here](https://github.com/nicsure/QuanshengDock/issues/86).

mikebanz, 2025-04-09:
Hi all, I'm having a similar problem. I'm wanting to be able to change the RX freq remotely on the radio, and I'd prefer to do it in python.

I've seen posts on threads about using a terminal program to just send the string (eg "FA045500000;") but those posts referred to CAT implemented on yaesu rigs.

I can happily recieiv the serial comms from the radio (and there certainly seems to be a lot of it...) but can't get the radio to respond to my attempts to talk to it.

The radio has the recommended firmware, and works fine with QD on a windows box, so I'm confident the setup is good, but... I kind of need to do it in python on a linux box.

Would anyone have a python snippet to set/change the frequency?

Much appreciated.

BranoSundancer, 2025-04-09:
This setup worked for me: SW that supports rigctl commands (was gpredict, could be Python) -tcp-> rigctld -CAT-virtual-serial-> QD -FTDI-serial-cable-> radio(with fw reqired by QD README)

If you want to connect Python directly to "FTDI-serial-cable" (and omit rigctld and QD/Windows), you could try to reverse-engineer the QD source and do all the stuff it does (in XVFO mode it takes-over the radio control to the PC, the display on radio will be blank then).

Or maybe, you could look at my funny approach [QuanshengPress](https://github.com/BranoSundancer/QuanshengPress) - it uses virtual key presses to change the frequency (works directly with FTDI serial port) and exposes **rigctld** interface for new frequency setting. It's not finished, never tested it on Linux, but could work for you.

mikebanz, 2025-04-09:
Hi Branano, thanks for the reply.

I was wondering about the keypress method, as there seemed to be 2 very different things going on in the QD software. (c sharp is not a language I'm familiar with, so sifting through the code is a bit painful.) For example, I'm trying to figure out how to put the radio into XVFO mode, although that seem more complicated than making a CAT call like FA0450.125; (which I can't get to work either...) The firmware documentation doen't cover what the radio is expecting and the comms flow  to provide the result.

In short, unfortunately I'm trying to avoid QD altogether.

My application is a radio in a remote station, connected to a linux box. I have a second system set up on the site with a hackrf, which is scanning the band and logging transmissions with time/date/freq/rssi. We need to relay the audio of the active channel to a rescue base over the internet from site. We are extremely limited for time and budget and only have the opportunity to get to site this weekend. (I knew about the problem a couple of months ago and wanted to work on it, but management didn't think it to be a priority until a recent bad rescue, so now its required immediately...) As a result I'm trying to hack together a solution with what I have to hand.

Currently there is a scanner cycling through the radio band, which takes 3-5 seconds. Unfortunately the band is dual-use and primarily for public comms, so occasionally some idle public chatter will halt the scanning as it breaks mute and remain until 5 seconds after the mute closes again. This means we can miss part or all of the distress call if it occurs during the chatting.

The Idea was for the logging setup to direct a 2nd radio to the strongest active channel in the band. I have used a 2nd hackrf on a linux box running a variant of the rtlsdr tool rtl_fm (called rx_rm), but the results are no fantastic in terms of recovered audio.

The other issue is that power at the site can be a problem. The linux boxes are able to reboot and recover after a loss, but the scanner needs several button presses to restart scanning. A windows box running qd would have similar issues from past experience.

In an ideal world, I would be able to write a script that would work something like: gotochannel.py 450.125 

Sorry about the massive detail, but I'm trying to cover as much detail as possible.

Cheers

BranoSundancer, 2025-04-11:
CAT commands you are refering are interpreted in QD. If you will not use QD/Windows, you will not have those CAT commands available. I am afraid the only way to achieve similar functionality without it is reverse engineering of the QD (at least).

But the QuanshengPress mentioned above could still work for you. It's more a toy than a production thing, but definitiely you can set a frequency from bash/python/anything and you don't need QD/Windows to make it work.

mikebanz, 2025-04-12:
Thanks for clarifying that. I kind of arrived near that conclusion, but didn't trust my interpretation of c #.

I ended up doing a bit of a re-write of you keypess solution, and it works well. It performs well in a test and is already deployed as a back-up for rescue comms.

Again, thanks for all your help