nicsure/QuanshengDock #86: [Documentation] serial protocol quick summary

unreal4me, 2024-11-14:
Hello there, very nice app and very nice use case!
It will be helpful if there is a semi documentation about the implementation of the CAT serial protocol. Like for example does it work with hamlib/rigctl etc?

BranoSundancer, 2024-11-14:
TL;DR: There is partial* emulation of **FT-991**, which is model **1035** in hamlib. This is the command I've been using with QD:
`rigctld.exe -m 1035 -r COM11 -s 38400 1 -vvvv`
The COM11 should be most probably set as `CAT Port` in QD's settings. And then it should work in the **XVFO mode**.

*The partial CAT implementation means not all CAT commands are emulated. Only those, which are required by **gpredict** connected to **rigctld**. See to PoC in the [Facebook group](https://www.facebook.com/groups/289656334131909/posts/364416183322590/). Maybe some other need to be added to support other programs.

BranoSundancer, 2024-11-14:
Here in [PTT.cs](https://github.com/nicsure/QuanshengDock/blob/103acd3f83ae0d920abfd38e0cd1ef242a9b8451/QuanshengDock/Serial/PTT.cs#L72)

unreal4me, 2024-11-14:
Thank you very much for your time Bruno, happy coding!

closed 2024-11-14 by unreal4me

BranoSundancer, 2025-07-18:
I am not 100% sure but I think I needed to create a virtual COM cable using [com0com](https://com0com.sourceforge.net/) utility. Rigctld was connected to COM11 (as seen above), QD was connected to virtual port COM10 (see screenshot below):

[image]

[image]