nicsure/QuanshengDock #89: Running the SW on a Raspi? Full remote control via Wifi

Bernie256, 2025-03-21:
This program is a cool idea. Nevertheless it has a major drawback: control of the radio is still by wire. If the radio is distant (e.g. outside the appartment on a balcony for better radio reception) you need to run many meters of cables through the appartment. How messy...
Usability would make a quantum leap if all this could be remote controlled - wireless! 
This could conceptually be implemented by letting the SW run on a Rasperry PI which has Wifi connectability. One could connect from the PC by VNC to the Raspi and remote-control it.

The Raspi + Radio + battery could be placed in a weather proof outside box., with an antenna connector and a power switch.
Mandatory: having a Linux Version of QuanshengDock to allow running it on a Rasperry PI. Any plans to do that?

BranoSundancer, 2025-03-22:
What about to use RPI as USB port redirector to your PC over a network?
See [Superuser post](https://superuser.com/questions/1720705/a-cross-platform-way-to-share-a-usb-device-over-a-network)

Bernie256, 2025-03-22:
Hmm, the post mentions USBIP. This would require some configuration on the Raspi, and also be installed and configured on the client Windows PC. One could see and try if that works, and if QuanshengDock reckognizes the remote Com Port.
Main problem: this would only provide the control interface. Audio links would not be available

BranoSundancer, 2025-03-22:
Good point, perhaps an external USB soundcard connected to RPI and another USBIP instance could propagate it to the PC as local. So at the end, two instances of USBIP would be needed (one for QD, second for audio). Personally I didn't test it, but it's worth of the experiment.